electrolama/zig-a-zig-ah #3: Help components eletronics

pedroborges1, 2020-02-11:
Good,  i am interested in this zigbee pen Zig-a-zig-ah.

you can send me a list of the electronic components you used ?

how do you flash this zigbee pen Zig-a-zig-ah ?

omerk, 2020-02-16:
Hi @pedroborges1, 

This is an untested, work-in-progress design. If you're interested in the BOM, the design files are in the repo and you can open them with EAGLE.

Keep an eye on [this Twitter thread](https://twitter.com/OmerK/status/1212864418155028480) for the latest info.

closed 2020-02-16 by omerk

pedroborges1, 2020-02-16:
> Hi @pedroborges1[3],
> Keep an eye on this Twitter thread[4] for the latest info.

thanks you ,

I already have the  zig-a-zig-ah[1]/ Revision A[2] eagle file .

but there is another folder name debug-adapter.

I just need the zig-a-zig-ah / revision A ?

the file is not there Bom ?

 

Ligações:
---------
[1] https://github.com/electrolama/zig-a-zig-ah
[2] https://github.com/electrolama/zig-a-zig-ah/tree/master/Revision%20A
[3] https://github.com/pedroborges1
[4] https://twitter.com/OmerK/status/1212864418155028480
[5]  
https://github.com/electrolama/zig-a-zig-ah/issues/3?email_source=notifications&email_token=AOQ2NEZLI2DCSCBALLOEEQLRDF3IVA5CNFSM4KS72SW2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEL4NJWI#issuecomment-586732761
[6]  
https://github.com/

omerk, 2020-02-16:
As I've said, this is still a work in progress so I have not yet released anything other than the currently untested design files. If you're interested in building one at this stage you are going to have to go through what is in the repo and figure it out yourself.

pedroborges1, 2020-02-16:
thank you ,when you are finished let me know .

 

Ligações:
---------
[1]  
https://github.com/electrolama/zig-a-zig-ah/issues/3?email_source=notifications&email_token=AOQ2NE76EO6ER2TVNGZLPY3RDG6BFA5CNFSM4KS72SW2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEL4UVKI#issuecomment-586762921
[2]  
https://github.com/

pedroborges1, 2020-02-18:
good, but have you tested this zig-a-zig-ah?

how did the flash zig-a-zig -ah ?

 

Ligações:
---------
[1]  
https://github.com/electrolama/zig-a-zig-ah/issues/3?email_source=notifications&email_token=AOQ2NE76EO6ER2TVNGZLPY3RDG6BFA5CNFSM4KS72SW2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEL4UVKI#issuecomment-586762921
[2]  
https://github.com/

pedroborges1, 2020-07-01:
good, i used an ikea repeater for the zigbee network, but even so a  
sensor sometimes stops working.

I think the signal doesn’t reach that sensor well, which is the one  
that’s farthest away.

some solution to improve this ?
 

Citando pedro <pedrofilipe32@sapo.pt>:

> thank you ,when you are finished let me know .
>
>  
>
>
>> As I've said, this is still a work in progress so I have not yet  
>> released anything other than the currently untested design files.  
>> If you're interested in building one at this stage you are going to  
>> have to go through what is in the repo and figure it out yourself.
>>
>
>  

 

Ligações:
---------
[1]  
https://github.com/electrolama/zig-a-zig-ah/issues/3?email_source=notifications&email_token=AOQ2NE76EO6ER2TVNGZLPY3RDG6BFA5CNFSM4KS72SW2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEL4UVKI#issuecomment-586762921
[2]  
https://github.com/