electrolama/zig-a-zig-ah #4: Raspberry hat

pimvanpelt, 2020-03-07:
Thanks for your work @omerk and I've subscribed to your mailing list for updates when you've tested your first Revision A in this repo.
When you're a bit further along, I suggest building a raspberry pi hat, as many folks will be using those to build their zigbee networks. I can contribute it (design it, order/test them and report back). Let me know if you'd be willing to accept that and I'll wait for you to confirm your current approach and send you a PR.
I think it would be good for the community to have these 26x2 boards in the same place and using the same firmware/pinout/etc.

omerk, 2020-03-08:
Hey @pimvanpelt, thanks for dropping by!

I've been busy testing today, you can find an update [on the project page](https://electrolama.com/projects/zig-a-zig-ah/).

With regards to the Raspberry Pi hat, not sure if you have seen [this](https://electrolama.com/projects/zoe/)? I have not yet pushed it online yet but I do have a version of zoe (zoe2, if you will) that merges the two projects together. Does that sound similar to what you had in mind?

pimvanpelt, 2020-03-09:
Yes, your zoe2 as a marriage of CC2652 and POE adapter looks like a
wonderful outcome to me, particularly because I do indeed use POE in my own
deployment.
Feel free to close this now, or once you have a prototype. Happy to buy a
dozen or so from you or make myself, as you prefer.

omerk, 2020-03-10:
Keep an eye on https://electrolama.com or [Subscribe to the mailing list](https://mailchi.mp/1746be86dd81/electrolama) to get updates when I publish new projects.

closed 2020-03-10 by omerk