electrolama/zig-a-zig-ah #5: [SUGGESTION] Team up with zigpy-cc and/or zigpy-znp projects to beta-test Zig-a-zig-ah with ZHA in Home Assistrant

Hedda, 2020-03-09:
@omerk Can I suggest that you try to reach out to @sanyatuning & @Adminiuga & @puddly about collaborating and teaming up to beta-test your "Zig-a-zig-ah!" USB-stick together with the zigpy-cc and/or zigpy-znp projects for using Texas Instruments CC2652 based Zigbee adapters with the ZHA integration in Home Assistant?

- https://github.com/sanyatuning/zigpy-cc/
- https://github.com/zha-ng/zigpy-znp
- https://github.com/puddly/zigpy-znp

As I understand it zig-a-zig-ah as well as zigpy-cc and zigpy-znp are only spare-time hobby projects but still I think that teaming up might give both projects some positive recognition in the Home Assistant community which might attract more developers and willing alpha/beta-testers from the community to come help out with both these projects which in the long run could benefit all users wanting ZHA to become more mature and working with more hardware than today.

- https://www.home-assistant.io/components/zha/

marv, 2020-03-09:
Great suggestion! I'd love to get involved as well if I can be of any help. I've been following both projects for the last week with great interest and been contemplating about ordering some zig-a-zig-ah boards myself to test things, but thought I wait for @omerk first test results. Now that the design is verified to work, i'll try to find time for that

puddly, 2020-03-09:
The [TI LAUNCHXL-CC26X2R1](https://www.zigbee2mqtt.io/information/supported_adapters.html#texas-instruments-cc26x2r1) uses the exact same SoC and people have had good luck with it using [zigbee2mqtt/zigpy-cc](https://github.com/Koenkk/zigbee2mqtt/issues/1429). It's been powering my entire network of 30+ devices for many weeks now so this adapter *should* work out-of-the-box when flashed with the same firmware.

Gamester17, 2020-03-10:
This kind of relate to my suggestion to donate money to existing Zigpy and ZHA developers for them to buy TI CC based hardware in order to be able to help with the development, see here:

https://github.com/zigpy/zigpy/issues/276

My thinking is if more developers with proven personal interest Zigpy or ZHA owed TI CC hardware then they might have a greater incentive to volunteer their time and skill for developing this.

I am not a developer myself, however, I am willing to make a small donation for this cause because believe that the concept of Zigpy or ZHA for Home Assistant will benefit the community in the long.

omerk, 2020-03-10:
Hey all,

Definitely keen to collaborate (power of open source hardware and software!) and can happily provide some prototypes if and when you have time to test it with your corresponding projects. 

I am reasonably confident with the design but will be doing some further testing this weekend and have enough parts for a couple of prototypes that I can hand populate/reflow. As you can imagine hand populating these boards take a good amount of time so the first few units are very precious 😄 

Feel free to reach out (<myfirstname>@electrolama.com) if you'd like to play with some prototype hardware!

Cheers,
Omer.

closed 2020-03-10 by omerk

Hedda, 2021-01-13:
@omerk Can I suggest that you reach out to @puddly and @sanyatuning  about prototype/alpha/beta/reference-testing your upcoming "zzh-p" (CC2652P USB) and "zoe2" (CC1352P shield) adapters together with zigpy? Please also see https://github.com/zigpy/zigpy-znp/issues/55