electrolama/zig-a-zig-ah #6: [SUGGESTION] Market and sell an initial of batch Zig-a-zig-ah! adapters via the Crowd Supply site?

Hedda, 2020-03-10:
@omerk I like to suggest that you market and sell an initial small batch of your Zig-a-zig-ah! USB-adapters via the Crowd Supply crowdsourcing site for electronics hardware at [crowdsupply.com](https://www.crowdsupply.com)

- https://www.crowdsupply.com

I have no personal affiliations to Crowd Supply but I backed a couple of projects there myself and know that their site is meant exactly for helping crowdsource funds and initial launch of a small batch of this type of early version hobby-class hardware kits meant for developers and advanced users who are willing to accept bleeding-edge hardware which might not yet be a finished product.

Personally I can recommend that you compare this μArt hardware by Karoly Pados which I backed:

- https://www.crowdsupply.com/pylo/muart

I think that his launch for a small initial batch of μArt (pronounced "Muart") on Crowd Supply was a perfect example of how a crowdsourcing campaign for this type of niche electronics hardware should be handled before its a mature product.

omerk, 2020-03-10:
Hey @Hedda, thanks for reaching out.

I am familiar with Crowdsupply and looking at distribution/logistics is on my list of things to do for the weekend, after I do some further testing with the prototypes so that I can issue files for manufacturing a small batch.

Getting the design verified is my current priority, rest will follow soon after 😄 

Cheers!

closed 2020-03-10 by omerk

Hedda, 2020-03-11:
@omerk Cool! Regardless if you go with that or not I still highly recommend that you read through μArt updates by Karoly Pados as a great example on how to run and update the community during such a campaign, so do checkout update history here https://www.crowdsupply.com/pylo/muart/updates