electrolama/zig-a-zig-ah #8: [REQUEST] Create a small adapter-board for connecting the standard CC-DEVPACK-DEBUG board to Zig-a-zig-ah USB-stick?

Hedda, 2020-04-01:
@omerk Can I suggest or request that you please consider also creating a matching small adapter-board for connecting Texas Instruments standard CC-DEVPACK-DEBUG board (XDS110 Debug Probe) to Zig-a-zig-ah USB stick? 

CC-DEVPACK-DEBUG board meaning the XDS110 Debug Probe that is also included in the CC2652 Launchpad board (LAUNCHXL-CC26X2R1):

- https://www.digikey.com/product-detail/en/texas-instruments/CC-DEVPACK-DEBUG/296-42039-ND/5232199
- https://processors.wiki.ti.com/index.php?title=CC13xx_CC26xx_Tools_Overview#XDS110_2

That is since the Zig-a-zig-ah USB-stick has a non-standard debug-port it would preferably be nice to have a simple adapter-board for connecting Texas Instruments standard CC-DEVPACK-DEBUG board to it for debugging purposes. Perhaps an adapter similar to that is commonly sold with CC2531 USB sniffer dongles for connecting a CC-Debugger to them, (those referred to as "downloader cable" by Zigbee2mqtt):

 - https://www.zigbee2mqtt.io/getting_started/what_do_i_need.html
 - https://www.zigbee2mqtt.io/getting_started/flashing_the_cc2531.html

For reference; @turboproc mentioned here that he has also designed/created a small adapter-board for connecting the standard CC-DEVPACK-DEBUG board to his own prototype CC2652 USB dongle:

https://github.com/Koenkk/zigbee2mqtt/issues/1429#issuecomment-606814358

turboproc, 2020-04-01:
Regarding the adapter, that's indeed what I did: tiny  board that clicks in the CC-DEVPACK-DEBUG board and has that connects with a small cable (2 x 3 pin 1.27mm pitch) into the dongle. Used the same pin-out on that side at he Zig-a-zig-ah to avoid again another non standard solution.

Hedda, 2020-04-01:
@turboproc Are you maybe willing to share your design layout & BOM, etc. as open-source hardware?

turboproc, 2020-04-01:
Sure, no problem. Give me some time to push it to Github.

omerk, 2020-04-01:
[image]

I've committed the files for this adapter [2 months ago](https://github.com/electrolama/zig-a-zig-ah/tree/master/Debug-Adapter), should have checked the repo @turboproc :-)

closed 2020-04-01 by omerk