electrolama/zig-a-zig-ah #9: [REQUEST] CAD-design of CC2652 USB-stick housing/enclosure and STL-files for editing and 3D-printing?

Hedda, 2020-04-07:
@omerk  Anyone working on an open-source CAD-design of a USB-stick housing/enclosure specifically made for CC2652 USB-sticks and if so would be willing to share their designs in STL file-format to enabling editing the design by others and 3D-printing?

One can, for example, get many designs and files for CC2531-based USB-dongles/USB-adapters (with or without antenna) on Thingiverse, example:

- https://www.thingiverse.com/thing:3948643
- https://www.thingiverse.com/thing:3646828
- https://www.thingiverse.com/thing:3532598
- https://www.thingiverse.com/thing:3731958

One can also buy professionally made injection moulding housings/enclosures from China via Alibaba or Aliexpress that one can take inspiration from, example:

- https://aliexpress.com/item/4000134201124.html
  - https://ae01.alicdn.com/kf/Hfb070d72fc45440497dce7d04e59d6b7d.jpg
- https://aliexpress.com/item/32855419632.html
  - https://ae01.alicdn.com/kf/H326a2ad3cfe64a318083102b0df62cd5N.jpg
  - https://ae01.alicdn.com/kf/H01eca8f47cd64bfd975738c1a4c30553i.jpg
- https://aliexpress.com/item/33007327270.html
- https://aliexpress.com/item/32852997963.html
- https://aliexpress.com/item/32690180597.html
- https://aliexpress.com/item/32847892743.html
  - https://ae01.alicdn.com/kf/HTB1K83IeXHM8KJjSZFwq6AibXXaD.jpg

By the way, I am not sure about English terminology for this, when talking about an exterior/outer protective covering for interior/inner electronic equipment in the specific context of a USB-dongle/USB-adapter like the Zig-a-zig-ah USB-stick, is the keyword called; enclosure, housing, casing, cover, case, box, shell, or another descriptive name that might be more correct? 

- https://en.wikipedia.org/wiki/Housing_(engineering)
- https://en.wikipedia.org/wiki/Electrical_enclosure
- https://en.wikipedia.org/wiki/Computer_case

Hedda, 2020-04-07:
Disclaimer: I have also posted the same request/question to @turboproc on https://github.com/turboproc/Zigbee-dongle-adapter/issues/1

omerk, 2020-05-07:
zzh uses a very common off-the-shelf "usb shell" enclosure from China so unfortunately I don't have any STLs to share as I didn't design it.

Happy to accept contributions from anyone with the CAD experience to release an open-source enclosure for it though, ping me if you need any dimensions/3D data exported.

closed 2020-05-07 by omerk