electrolama/zig-a-zig-ah #11: [QUESTIONS] Plans on manufacturing, crowdfunding and/or selling none-prototype version of Zig-a-zig-ah (ZZH) dongles?

Hedda, 2020-05-07:
@omerk Read in https://github.com/Koenkk/zigbee2mqtt/issues/1429 that you plan on setting up a Tindie store for Zig-a-zig-ah (ZZH) dongles and was curious on your plans for selling and shipping the USB-stick adapter as a "finished plug-and-play product" there as not seen any updates on your Twitter account or the Electrolama site. 

Mostly I was wondering if you already have funding and manufacturers lined up for making a finished product or are you doing a crowdfunding campaign? Also, where do you think you will be shipping it from? I'm asking for myself as an EU (European Union) customer, Sweden to be exact.

Regardless of how you proceed, please ship the product with a nice plastic housing/enclosure/shell/chassis which will be easy to open and close for modification, as related to the request in https://github.com/electrolama/zig-a-zig-ah/issues/9 you can buy professionally made injection moulding housings/enclosures from China but I am not sure they are to open and close over and over again without breaking the fastenings.

zigastegu, 2020-05-07:
Onboard bootloader sholod enable flashing new fw without opening enclosure or so I have read on: https://electrolama.com/projects/zig-a-zig-ah/

Do you have any other use cases for opening enclosure?

omerk, 2020-05-07:
Hi @Hedda,

Not interested in crowdfunding, the first batch should go up on Tindie at some point this weekend (at least, submitted for their approval). Sign up to the [mailing list](https://electrolama.com/list) to get an update when everything goes live.

Here's a photo of the enclosure and the antenna that will ship with the units purchased on Tindie, it's a snug fit case but certainly would last the few times that you might need to open the unit if you want to program the device

[image]

omerk, 2020-05-07:
@zigastegu If you want to program the board, you will still need to press the "BSL" button as you're plugging the device in to trigger the bootloader (requiring you to open the case). 

I've asked about triggering the bootloader from a running program on the [TI forums](https://e2e.ti.com/support/wireless-connectivity/zigbee-and-thread/f/158/t/902829), let's see what they say.

Realistically, unless you are a firmware developer you shouldn't have to worry about opening the case to access the button as once you program the board and set up your network, you will leave things as they are. If you are indeed developing firmware and don't want to worry about the case, I suggest using the bare board without the case.

mar565, 2020-05-07:
@omerk can you already Tell us the price?

omerk, 2020-05-10:
Listing submitted to Tindie, once approved it will go live on https://www.tindie.com/products/electrolama/zzh-cc2652r-multiprotocol-rf-stick/

closed 2020-05-10 by omerk

Hedda, 2020-05-11:
> Onboard bootloader should enable flashing new fw without opening enclosure or so I have read on: https://electrolama.com/projects/zig-a-zig-ah/
> 
> Do you have any other use cases for opening enclosure?

Firmware developers with the debug adapter require access to the board for debugging, see:

https://github.com/electrolama/zig-a-zig-ah/tree/master/Debug-Adapter

Also, if you brick it then you need to press the "BSL" button to trigger boot to the bootloader.

Hedda, 2020-05-12:
@omerk Out of stock / Sold out already? Checked yesterday before went to bed and was not live then:

https://www.tindie.com/products/electrolama/zzh-cc2652r-multiprotocol-rf-stick/