electrolama/zig-a-zig-ah #12: Some suggestions for a future revision of the board

puddly, 2020-05-09:
(copied from https://github.com/zha-ng/zigpy-znp/issues/10)

Thanks for the awesome board! I'm seeing an average LQI improvement of 30% compared to the bulkier LAUNCHXL-CC26X2R1.

Here are some suggestions for future revisions of the board:

1. Would you consider using a USB-serial chip by FTDI or Silicon Labs? The CH340 is the only USB-serial chip that consistently causes a kernel panic on my MacBook and is overall unreliable in my experiences with random dev kits. I'm running kernel 4.14.176 on my ODROID XU4 and the ZZH stick needs to be re-plugged a few times in a row to respond to any serial communications, unless it's plugged into a USB extender (!?). I have half a dozen other USB devices plugged in at any given time and have never had this problem except with devices using the CH340.

2. While a minor detail, holding down the flash button while inserting the stick is difficult with the button mounted horizontally and near the male USB-A connector. It would be a bit easier to flash if the button was mounted near the rear of the board next to the antenna.

3. I'm not sure if there's a huge price difference for you but including one of those bendable and rotatable 2.4Ghz antennas would make it easier to mount. Most of my USB devices are plugged into a powered hub with the ports sticking straight up and the fixed right-angle antenna connector makes mounting a little difficult.

If I end up making a case for it I'll definitely make a pull request with the STL. If you have a moment, could you upload an SVG version of your logo to this repository?

omerk, 2020-05-10:
Hey @puddly, great to hear you are up and running and thanks for the comments!

Addressing your points:

1) zzh originally had CP2102N but had to find a replacement as it simply wouldn't fit the layout once I picked the enclosure. I did look at FT230X and FT234XD (the 'runner-up') but at many times the cost of CH340E and with much better driver support over the last year or so for CH340E, the decision was made. 

The assumption here is that majority of the users of zzh will be using Linux and probably on a popular SBC (like a Raspberry Pi) that will have an up-to-date kernel. In the many months of testing I did with both Raspberry Pi and an Ubuntu box, I never had driver issues. I know @Koenkk had an issue initially with the drivers but that was solved with him updating his box.

Now of course this isn't an attempt at not acknowledging the issues you are having, just providing the rationale behind the design decision. For macOS, unfortunately I can't really comment as I don't have any Macs around but for your XU4, can I ask perhaps you give a more recent kernel (if available) a go? [Some recent fixes in the kernel driver](https://github.com/torvalds/linux/commits/master/drivers/usb/serial/ch341.c) did improve the situation quite a bit. If you have the willpower and the time to tinker with the kernel on your XU4, perhaps we can get to the bottom of it? In the meantime, I can happily send you the hacked-with-the-FT234XD board I have in my box of prototypes. It won't look pretty but it sure is another collectors' piece :-)

2) I ordered a dozen or so button samples, some side mounted ones included, but it came down to picking this current one because it had a nice click and actually felt like a button (a problem with tiny buttons!). I actually wanted to use the side mounted one used on the TI Launchpads but not enough space :(

3) The antenna I picked sits nicely in the middle of a size-cost-performance triangle, size being the deciding factor. I couldn't find a small-enough 'bendable' antenna so decided to go for this fixed version instead, based on the assumption that the following would be the typical use case:

[image]

I am still looking for a small *and* bendable antenna that is sensibly priced, once I find it we can then issue it as a running change for boards shipped. The beauty of the detachable antenna is that if it gets on your nerves too much, you can change it.

Always difficult defining a product to satisfy all use cases and it comes down to making assumptions (like the ones discussed above) and hoping you were right. I guess we will find out given that this is an open source project with an issue tracker :-)

Cheers!

Adminiuga, 2020-05-10:
well, since there's SMA connector you could get any antenna you'd like aliexpress. Just make sure it is SMA Male connector for antenna. 

eg https://www.aliexpress.com/item/33044689298.html

Adminiuga, 2020-05-11:
@puddly are you on armbian? I'm running my prod on xu4 as well and this is going be a show stopper. Gonna try switching kernels to 5.4

puddly, 2020-05-11:
1. My development and production setup is definitely not common so no worries, I'm just documenting the issues I have. If I run into it again and figure out a way to fix it I'll post an update.

2. I was thinking just moving the button closer to the antenna, so you can keep it pressed more easily while inserting the stick. It does have pretty good tactile feedback for its size.

3. I just tested out one of those bendable antennas from a 2.4GHz wireless card and the LQI was reduced quite a bit, though not as much as with the right-angle antenna pointed the wrong way.

Hedda, 2020-05-11:
Possible to replace CH340E with one that has flashable EEPROM so strings/VID/PID can be customized?

Would be both more user-friendly, to be able to for example script plug-and-play auto-detection, and more professional if the description for the serial port device was shown as example:

 "Zig-a-zig-ah USB-dongle v1.0.0"

As per the original suggested in https://github.com/zha-ng/zigpy-znp/issues/10 by @Adminiuga 

> @omerk is it possible to customize the USB-Serial chip, so it provides a more specific description for serial port device? Currently I"m getting generic:
> `usb-1a86_USB2.0-Serial-if00-port0 -> ../../ttyUSB0`

Hedda, 2020-05-11:
> If I end up making a case for it I'll definitely make a pull request with the STL. If you have a moment, could you upload an SVG version of your logo to this repository?

@puddly Please also see related CAD-design request in https://github.com/electrolama/zig-a-zig-ah/issues/9

Hedda, 2020-05-11:
> While a minor detail, holding down the flash button while inserting the stick is difficult with the button mounted horizontally and near the male USB-A connector. It would be a bit easier to flash if the button was mounted near the rear of the board next to the antenna.

@omerk Regardless of button location on the board, a tip follow the industry-standard design and drill a small hole through the enclosure/case where the button is located to let users be able to press the button with a paper-clip

https://www.google.com/search?q=paper+clip+reset+button&tbm=isch

Again, having a "Reset Hole" purposely design to be pressed via a paper-clip is the industry-standard.

Hedda, 2020-05-11:
SMA or RP-SMA (Reverse-Polarity SMA) is basic but important information when buying an antenna:

https://en.wikipedia.org/wiki/SMA_connector

@omerk Which exact type of SMA connector design does the ZZH board have on it? 

Do ZZH board have a Standard-Polarity SMA or a Reverse-Polarity ("RP") SMA female connector?

> well, since there's SMA connector you could get any antenna you'd like aliexpress. Just make sure it is SMA Male connector for antenna.
> 
> eg https://www.aliexpress.com/item/33044689298.html

I can personally recommend "Delock" branded Wi-Fi antennas which can be bought in EU & USA.

https://www.delock.com/produkte/G_1468_Antenna.html

I have no personal affiliations but in the European Union you can buy Delock antennas from Conrad

https://www.conrad.com

puddly, 2020-05-11:
Another issue I'm seeing (though probably not zzh-specific) is that I get nonstop routing and MAC failures with it plugged directly into my XU4. There are bulbs within a few feet of it I get errors such as these in response to any request, even with route discovery enabled:

```Python
AF.DataConfirm.Callback(Status=<Status.NwkNoRoute: 205>, Endpoint=1, TSN=18)
AF.DataConfirm.Callback(Status=<Status.MacChannelAccessFailure: 225>, Endpoint=1, TSN=15)
```
Plugging it into a hub an inch or two away fixes the issue. I'll try out some simple shielding and will see if it helps at all.

Adminiuga, 2020-05-11:
ChannnelAccessFailure  is pretty much a tail sign of RF interference. If it is fixed by keeping the board away from SBC, then we can conclude the noise is coming from the  SBC which is frankly quite expected.  ConBees are susceptible  to the very same issue and that's why they always recommend to use an usb extension cable.
I wonder how many people complaining about range issue, could fix them  by using an usb extension.  I assume all are susceptible to this issue, just some  adapters are not providing this feedback.

Hedda, 2020-05-12:
Tip regardless of antenna is to use at least a 1-meter USB-extension cable for all radio adapters. If use USB-extension cable then gets it away from most SBC or computers WiFi and Bluetooth interference.

Hedda, 2020-05-12:
@omerk here is a "must-read" article about analyzing antennas if want to be sure before deciding:

- https://tinkerman.cat/post/analyze-your-antennas-using-an-aai-n1201sa/

It recommends buying theN1201SA by Accuracy Agilty as is a cheap but incredibly useful tool:

- http://s.click.aliexpress.com/e/bVPlgjT6
- https://amzn.to/2VqBlW5

That article is written by Xose Pérez (a.k.a. Tinkerman) who also develops ESPurna FW for ESP8266:

> **Conclusions**
> Being able to profile an antenna is a must-have when you are dealing with RF. The N1201SA is an incredibly useful tool for the price it has and one of the best acquisitions I've ever done. If you or your community have the chance to get one to choose the proper antennas for your sensors or the receiver, do not doubt about it.
> 
> And about the antennas. Performance heavily depends on manufacturing procedure but not only. The values I got from similar antennas have been sometimes very different. That reinforces the utility of tools like the N1201SA since you will probably want to test the specific antenna you plan to deploy. A simple antenna analyzer is the first filter, next would be to measure how much power they actually radiate and in what directions. Do they really have the gain their manufacturers claim… I'd be surprised if those small whip 868MHz antennas really get the 5dBi. Let's see…

sjorge, 2020-06-27:
> 1. Would you consider using a USB-serial chip by FTDI or Silicon Labs? The CH340 is the only USB-serial chip that consistently causes a kernel panic on my MacBook and is overall unreliable in my [...]

Just going to give this a +1 here too :) Omerk is already aware of my preference, as I am not using linux (or freebsd) I am currently trying to write a driver for CH340 and boy its and oddball chip.

omerk, 2020-06-27:
zzh-p will use FT231 that will allow us customise strings and also auto-reset/put device in BSL mode via DTR/RTS pins exposed.

No plans for change on zzh though.

copystring, 2020-06-27:
Nice. I was about to ask about this to allow nice udev rules (all CH340 have the same serial). Exciting!

svh1985, 2020-09-18:
I also have a small suggestion;
I'm not sure if the provided plastic case is custom designed or off-the-shelf, but adding a small hole to allow us to press the BSL button would be very nice. Getting the case off is hard and this makes programming possible without removing the case.

Hedda, 2021-01-13:
@omerk Any news on your CC2652P based "zzh-p" and if it will soon be available for sale or not?

omerk, 2021-01-22:
Hi all, lots of comments to address but as a summary:

There will be two versions zzh with CC2652P: zzhp and zzhp-lite. Both will use the same firmware, zzhp-lite will be module-based (PCB antenna) and zzhp will be a custom board similar to the first draft screenshot you linked above (external antenna). I've also managed to squeeze in a few more goodies in zzhp and will be updating the website soon. Release date not set yet but it's very soon.

@svh1985 and many others requested a pre-drilled hole. The last couple of (and all future) batches will ship with cases that are pre-drilled:

[image]

You'll also notice that we now have a swivel antenna, another requested feature.

Both drilled cases and that newer antenna were things I wanted from the get go but needed to hit a certain MOQ before I could get them. We're there now 🙂

closed 2021-01-22 by omerk