electrolama/zig-a-zig-ah #15: No Signal Z2M when using a different antenna

kennymc-c, 2020-06-28:
When I connect the zzh to different antenna (dual band WiFi +9dbi), Z2M seems to have problems with reception. There are simply no signals coming in. According to dmesg, the adapter has been detected by the system. As soon as I reconnect it to the supplied antenna, the signal is coming back. Since I use the same type of antenna with a CC2530 router, I assume that the antenna itself is suitable for this. I have attached a zigbee-herdsmann debug log that might help:

https://pastebin.com/k4wAtciK

puddly, 2020-06-28:
The zzh has a female SMA connector. Make sure your antenna is not a male RP-SMA, which will still screw on but will not actually act as an antenna:

[image]

kennymc-c, 2020-06-28:
Thanks, that should be the reason. My antenna is also female. I just realized now when I saw the connection. Sorry, I could have figured it out myself.

closed 2020-06-28 by kennymc-c