electrolama/zig-a-zig-ah #16: [SUGGESTION] Team up with pipiche38 who is developing a zigpy plugin for Domoticz

Hedda, 2020-08-25:
@omerk (Omer Kilic) Can I suggest that you and @electrolama try to reach out to @pipiche38 (Patrick Pichon) about possible collaborating and teaming up to test your "Zig-a-zig-ah!" USB-stick together with the new Domoticz implementation that he is is developing using [zigpy](https://github.com/zigpy/zigpy)?

* https://github.com/pipiche38/Domoticz-Zigpy/

If omerk /Electrolama could donate one or two of your "Zig-a-zig-ah!" USB-sticks to him then maybe he that can help @pipiche38 developing effort as it would give him the possibility to also test both [zigpy-cc](https://github.com/zigpy/zigpy-cc) and [zigpy-znp](https://github.com/zha-ng/zigpy-znp) radio libraries for zigpy to try out their compatibility with newer Texas Instruments CC2652 based Zigbee adapters as zigpy can support many different adapters.

https://github.com/zigpy/zigpy

pipiche38 is still in a very early stage of developing his "Domoticz-Zigpy" plugin for [Domoticz](https://www.domoticz.com) (a free and open-source home automation software), I understand in fact the project is so a such early in development that is not yet usable, however, I read on Domoticz community forum that he currently only has access to ZiGate Zigbee adapters (this as pipiche38 is also the developer of an existing ZiGate plugin for Domoticz). So at least we know that pipiche38 already has the experience and skills to develop Zigbee plugins for Domoticz.

* https://github.com/pipiche38/Domoticz-Zigate-Wiki/blob/master/en-eng/Home.md

* https://www.domoticz.com/wiki/Zigate

As I understand it zig-a-zig-ah by Electrolama, zigpy, Domoticz, and this new "Domoticz-Zigpy" plugin are all only spare-time hobby projects but still, I think that teaming up might give all involved projects some positive recognition in the Domoticz community and the whole home automation scene in general which might attract more developers and willing alpha/beta-testers from the community to come to help out with all these involved projects which in the long run could benefit all users wanting Zigbee in Domoticz to become more mature and working with more Zigbee hardware dongles than it is today.

omerk, 2020-08-27:
@pipiche38 Happy to support a fellow open-source Zigbee developer, if you would like me to send you a zzh stick please drop me an email (omer @ electrolama dot com) with your address.

closed 2020-08-27 by omerk

pipiche38, 2021-12-14:
@omerk this is just to let you know that we have restarted the project to integrate Electrolama zzh key into Domoticz.
This will be done over the Domoticz-Zigate for which we will provide a zigpy layer and which will provide the support.

We will try to track the work over the Issue  [#888](https://github.com/pipiche38/Domoticz-Zigate/issues/888)

pipiche38, 2021-12-19:
@omerk just a quick update from our side, we are making very good progress on the zigpy-znp integration into the Domoticz-Zigate plugin

The main benefit is by using the zigpy-znp layer on top of the zzh key is to use the all development done already with Zigate and so getting all feature available right away.

we are now able to get the Domoticz (zigate) plugin connected to a zzh key, controlling the key ( seting permit to join, managing the Led ...) and capable to pair Aqara sensor as Bulbs from Ikea or LXEK.

Struggling with the ZDO zigpy-znp layer for now, but no real big issue

Will probably be join by one more developper/tester/documentation soon @jp_keros

Hedda, 2021-12-20:
> Struggling with the ZDO zigpy-znp layer for now, but no real big issue

@pipiche38  Off-topic but FYI check out related discussion with reply from puddly regarding ZDO -> https://github.com/zigpy/zigpy-znp/issues/108

jp-keros, 2021-12-20:
> Will probably be join by one more developper/tester/documentation soon @jp_keros

I'm involved.

pipiche38, 2021-12-27:
@omerk , just a quick update. 

Project is progressing fast, here is the future home page: [ZigbeeforDomoticz](https://github.com/pipiche38/Domoticz-Zigate/tree/zigpy#readme) feel free to comment, update.

I there are some zzh users interested to be part of a testing community , please point them to me, we will for sure be interested to have a panel of end user testers.