electrolama/zig-a-zig-ah #19: [SUGGESTION] Team up with zoic21 who is developing ZGP support for zigpy

Hedda, 2020-11-18:
@omerk (Omer Kilic) Can I suggest that you (and @electrolama ) try to reach out to @zoic21 (Loïc Gevrey) about possible collaborating and teaming up to test your "Zig-a-zig-ah!" USB-stick together with the ZGP (Zigbee Green Power) implementation that he is developing for [zigpy](https://github.com/zigpy/zigpy) to be used in Home Assistant and Jeedom?

https://github.com/zigpy/zigpy/pull/519

If you omerk (Electrolama) could donate one or two your "Zig-a-zig-ah!" USB-sticks to him then maybe he that can help @zoic21 developing effort as it would give him the possibility to test a TI CC2652 based Zigbee adapter like your with ZGP (Zigbee Green Power) implementation with the [zigpy-znp](https://github.com/zha-ng/zigpy-znp) radio libraries for Texas Instruments Z-Stack 3 adapters. It could perhaps later even be used as a reference adapter. No strings attached of course as these are still hobby projects at their core with only one developer so far. 

https://github.com/zigpy/zigpy

As I understand it, today @zoic21 (Loïc Gevrey) only have a Silicon Labs based Zigbee adapter plus a ConBeee Zigbee adapter to test this implementation with and apparently, it requires protocol translation from each propitiatory API that different hardware manufacturers use on their Zigbee adapters.

PS: @zoic21 (Loïc Gevrey) also happen to be the lead developer of Jeedom (open-source home automation software from France)

zoic21, 2020-11-18:
Hello,
Yes I only have conbee 2 (deconz key) en silicon labs key. I test ZGP on it and it's seem ok but I don't know how it's work on TI. 
From what I saw on deconz and silicon labs there is a bit of decoding done by the key and for each key it is different.

Hedda, 2020-11-19:
@zoic21 this is only a request for Omer to donate a TI dongle hardware to you in order so that you can develop with it. 

For the actual development of ZGP for TI in zigpy you will need to talk to zigpy-znp developers in https://github.com/zigpy/zigpy-znp/issues/47

th0m4sek, 2020-11-19:
I believe in yours good intentions but in my opinion this is not the way to ask for this kind of staff.
You can buy one and send it @zoic21 or donate money directly so he will buy it by himself.
I don't want to be disrespectful to anybody.
Peace.

Hedda, 2020-11-19:
@th0m4sek I disagree as the end result should benefit Electrolama/omerk.  Think of it based on these facts and extrapolations:

Electrolama/omerk sells his zig-a-zig-ah adapter on Tindie today. If the zigpy-znp library supports ZGP then we can assume that zig-a-zig-ah adapter will soon be compatible with ZGP in Home Assistant and Jeedom. News of ZGP support with zig-a-zig-ah adapter will probably mean that Electrolama/omerk will sell more adapters on Tindie. If however zigpy only get support for ZGP with ConBee and Silicon Labs dongles then we can assume that at least some potential customers will instead buy those because they have better support, instead of buying zig-a-zig-ah adapter.

So in my opinion it does not to ask for a small hardware donation on behalf of a developer than is working on something like this.

PS: You should know that Electrolama/omerk has already donated zzh hardware donles to other zigpy-znp developers which directly or indirectly have lead to it becoming one of the best radio libraries for zigpy (which is used for Zigbee in Home Assistant), see https://github.com/electrolama/zig-a-zig-ah/issues/5

PPS: Disclaimer; I am just a fan of the zigpy project, I already own both a ConBee and a Silicon Labs dongle so have myself no personal gains doirectly from zigpy-znp getting ZGP support, but  I believe that ZGP support for all zigpy compatible hardware will indirectly benifit the whole community in the long-term as better Zigbee support in open source home automation software means more users which in turn normally leads more competition = less expensive devices = more developers = better products for all.

omerk, 2021-01-22:
Hi all,

To support the open-source home automation development community, I am happy to send boards to active developers of open-source home automation projects, costs covered by the Electrolama operation. Don't have the bandwidth to monitor Github issues spread across repos though so please get in touch (support@electrolama.com) with a link and a summary of your project and stock levels allowing I will ship you a board or two, or prototypes of upcoming projects.

closed 2021-01-22 by omerk