electrolama/zig-a-zig-ah #21: Failed to probe ZNP radio

Zaakupendrak, 2020-12-16:
Hi.

I got problem to create zigbee integration with zzh! under Home Assistant.
**HA image**: raspberrypi4-64-homeassistant:2020.12.0 
**Platform**: Debian 10
**stick**: zzh!
**connection_type**: wire

[image]

`root@rpi4-20201112:~# docker logs homeassistant
2020-12-16 10:28:43 WARNING (MainThread) [zigpy_zigate.api] No response to command 0x0002
2020-12-16 10:28:43 WARNING (MainThread) [zigpy_zigate.api] Retry command 0x0002
2020-12-16 10:28:46 WARNING (MainThread) [zigpy_xbee.api] at: No response to AP command
2020-12-16 10:28:47 WARNING (MainThread) [zigpy_cc.api] No response for: SRSP SYS version {}
2020-12-16 10:29:39 WARNING (MainThread) [zigpy_znp.zigbee.application] Failed to probe ZNP radio with config {'path': '/dev/serial/by-id/usb-1a86_USB_Serial-if00-port0', 'baudrate': 115200, 'flow_control': None}
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/zigpy_znp/zigbee/application.py", line 167, in probe
    await znp.connect()
  File "/usr/local/lib/python3.8/site-packages/zigpy_znp/api.py", line 234, in connect
    self._version = await self.request(c.SYS.Version.Req())
  File "/usr/local/lib/python3.8/site-packages/zigpy_znp/api.py", line 507, in request
    response = await response_future
asyncio.exceptions.CancelledError
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/zigpy_znp/zigbee/application.py", line 167, in probe
    await znp.connect()
  File "/usr/local/lib/python3.8/site-packages/async_timeout/__init__.py", line 55, in __aexit__
    self._do_exit(exc_type)
  File "/usr/local/lib/python3.8/site-packages/async_timeout/__init__.py", line 92, in _do_exit
    raise asyncio.TimeoutError
asyncio.exceptions.TimeoutError
`

`root@rpi4-20201112:~# docker ps  
CONTAINER ID   IMAGE                                                   COMMAND                  CREATED          STATUS          PORTS                  NAMES
cc4c253552c9   homeassistant/raspberrypi4-64-homeassistant:2020.12.0   "/init"                  13 minutes ago   Up 13 minutes                          homeassistant
ee0bb0f6fe52   homeassistant/aarch64-hassio-multicast:3                "/init"                  13 minutes ago   Up 13 minutes                          hassio_multicast
cbbc7d439c70   homeassistant/aarch64-hassio-observer:2020.10.1         "/init"                  13 minutes ago   Up 13 minutes   0.0.0.0:4357->80/tcp   hassio_observer
c61debfb4211   homeassistant/aarch64-hassio-cli:2020.11.1              "/init /bin/bash -c …"   13 minutes ago   Up 13 minutes                          hassio_cli
26ded078ee16   homeassistant/aarch64-hassio-audio:17                   "/init"                  13 minutes ago   Up 13 minutes                          hassio_audio
fba6e4c7b005   homeassistant/aarch64-hassio-dns:2020.11.0              "/init"                  14 minutes ago   Up 14 minutes                          hassio_dns
5d250adf5551   homeassistant/aarch64-hassio-supervisor                 "/init"                  14 minutes ago   Up 14 minutes                          hassio_supervisor
`

`pi@rpi4-20201112:~$ dmesg | grep -A 5 -B 5 341
[    9.339521] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[    9.346715] usbhid: USB HID core driver
[    9.365962] input: HOLDCHIP USB Gaming Keyboard as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.1/1-1.1:1.0/0003:04D9:0024.0001/input/input0
[    9.372458] platform regulatory.0: firmware: failed to load regulatory.db (-2)
[    9.395476] firmware_class: See https://wiki.debian.org/Firmware for information about missing firmware
[    9.397257] usbcore: registered new interface driver ch341
[    9.407699] platform regulatory.0: Direct firmware load for regulatory.db failed with error -2
[    9.417950] usbserial: USB Serial support registered for ch341-uart
[    9.419898] videodev: Linux video capture interface: v2.00
[    9.427286] cfg80211: failed to load regulatory.db
[    9.464429] hid-generic 0003:04D9:0024.0001: input,hidraw0: USB HID v1.10 Keyboard [HOLDCHIP USB Gaming Keyboard] on usb-0000:01:00.0-1.1/input0
[    9.471746] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    9.492706] usbcore: registered new interface driver brcmfmac
[    9.494774] ch341 1-1.4:1.0: ch341-uart converter detected
[    9.495750] input: HOLDCHIP USB Gaming Keyboard System Control as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.1/1-1.1:1.1/0003:04D9:0024.0002/input/input1
[    9.532405] brcmfmac mmc0:0001:1: firmware: direct-loading firmware brcm/brcmfmac43455-sdio.bin
[    9.539173] usb 1-1.4: ch341-uart converter now attached to ttyUSB0
[    9.554597] input: HOLDCHIP USB Gaming Keyboard Consumer Control as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.1/1-1.1:1.1/0003:04D9:0024.0002/input/input2
[    9.575740] brcmfmac mmc0:0001:1: firmware: direct-loading firmware brcm/brcmfmac43455-sdio.raspberrypi,4-model-b.txt
[    9.588074] snd_bcm2835: module is from the staging directory, the quality is unknown, you have been warned.
[    9.605654] bcm2835_v4l2: module is from the staging directory, the quality is unknown, you have been warned.
[    9.605659] input: HOLDCHIP USB Gaming Keyboard as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.1/1-1.1:1.1/0003:04D9:0024.0002/input/input3
--
[  995.836734] IPv6: ADDRCONF(NETDEV_CHANGE): veth19e4479: link becomes ready
[  995.849219] hassio: port 5(veth19e4479) entered blocking state
[  995.860442] hassio: port 5(veth19e4479) entered forwarding state
[ 1013.847766] udevd[190]: starting version 3.2.9
[ 1013.859926] udevd[191]: starting eudev-3.2.9
[ 1093.394825] ch341-uart ttyUSB0: usb_serial_generic_read_bulk_callback - urb stopped: -32
[ 1093.409063] ch341-uart ttyUSB0: usb_serial_generic_read_bulk_callback - urb stopped: -32
[ 1093.529300] usb 1-1.4: USB disconnect, device number 4
[ 1093.539899] usb 1-1.4: ch341_read_int_callback - usb_submit_urb failed: -19
[ 1093.552792] usb 1-1.4: failed to send control message: -19
[ 1093.564326] ch341-uart ttyUSB0: ch341-uart converter now disconnected from ttyUSB0
[ 1093.575456] ch341 1-1.4:1.0: device disconnected
[ 1097.336993] usb 1-1.3: new full-speed USB device number 5 using xhci_hcd
[ 1097.455084] usb 1-1.3: New USB device found, idVendor=1a86, idProduct=7523, bcdDevice= 2.64
[ 1097.469001] usb 1-1.3: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[ 1097.479738] usb 1-1.3: Product: USB Serial
[ 1097.493797] ch341 1-1.3:1.0: ch341-uart converter detected
[ 1097.508005] usb 1-1.3: ch341-uart converter now attached to ttyUSB0
`

Changing port/repluging/restarting RPI does not help.

Hedda, 2021-01-13:
@Zaakupendrak if this was an issue with ZHA in Home Assistant or the zigpy-znp library then you might have been better of posting to https://github.com/home-assistant/core/ and/or https://github.com/zigpy/zigpy-znp however I am wondering if you have really managed to correctly flash an application firmware the adapter properly with Z-Stack 3 Zigbee stack application firmware as zzh does ship pre-flashed with a Zigbee coordinator firmware.

https://electrolama.com/projects/zig-a-zig-ah/

"**_Please keep in mind that zzh is a general purpose development board and as such, it is shipped "blank" with no code on the wireless microcontroller. You will need to program it before it does anything meaningful._** _Given the application dependent nature of this board, limited after-sales support can be provided for Tindie purchases (For example: we can help you program this board but can't fix your Zigbee network range issues or offer specific software support)._"

Suggest that you install drivers, enter bootloader mode and try using BSL again to flash correct firmware Z-Stack coordinator firmware from @Koenkk's firmware repository. 

https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.x.0/bin

"**_Download the Z-Stack coordinator firmware from @Koenkk's firmware repository. The firmware you'll need can be found under coordinator/Z-Stack_3.x.0/bin/CC26X2R1_<date>.zip, **_as of writing the latest version available is CC26X2R1_20200805.zip. Download and extract this and follow the "Flashing using BSL" instructions to burn this on your zzh_**_**."

omerk, 2021-01-22:
@Zaakupendrak hope you were able to solve your problem.  I don't monitor Github issues for support, please send an email to support@electrolama.com instead if you'd like some help (it does seem like it is either a firmware or a ZNP configuration issue there).

closed 2021-01-22 by omerk

Zaakupendrak, 2021-01-22:
Hi,

The problem was that I didn’t flash stick. Once I did, everything seems
working :)

Thank you for your attention.

Hedda, 2021-01-22:
> Hi, The problem was that I didn’t flash stick. Once I did, everything seems working :) 

Suspected it was so. Suggested in https://github.com/electrolama/zig-a-zig-ah/issues that future zzh sticks shipped pre-flashed.