electrolama/zig-a-zig-ah #22: Timeout waiting for ACK/NACK after 'Synch (0x55 0x55)'

seb-koch, 2020-12-24:
Hi,

i am seeking help as i am running out of ideas. I ordered two zzh and i am unable to flash the koenk firmware to either of them. I tried several procedurs, including flashing a blink.bin to the device, erasing flash and such. I followed all the instructions and always unplugged the stick, held the bsl button, plugged it back in (waited for 3, 5 or 10 seconds) and tried flashing to it. None of it worked so i guess i am doing smth really wrong here or i am missing an important step.

Here is the output after plugging it back in with BSL (blinking stopped so the blink.bin wasn't loaded i guess).
```
sebo@esmombibi:~/download/cc2538-bsl-master$ sudo ./cc2538-bsl.py -p /dev/ttyUSB0 -evw CC26X2R1_20201026.hex
Opening port /dev/ttyUSB0, baud 500000
Reading data from CC26X2R1_20201026.hex
Your firmware looks like an Intel Hex file
Connecting to target...
ERROR: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
sebo@esmombibi:~/download/cc2538-bsl-master$ sudo ./cc2538-bsl.py -p /dev/ttyUSB0 -evw CC26X2R1_20201026.hex
Opening port /dev/ttyUSB0, baud 500000
Reading data from CC26X2R1_20201026.hex
Your firmware looks like an Intel Hex file
Connecting to target...
ERROR: Timeout waiting for ACK/NACK after 'Synch (0x55 0x55)'
```

And here is the dmesg showing it was detected.
```
[   92.467388] ch341 1-1:1.0: ch341-uart converter detected
[   92.467772] usb 1-1: ch341-uart converter now attached to ttyUSB0
[  146.754652] usb 1-1: USB disconnect, device number 6
[  146.754836] usb 1-1: failed to send control message: -19
[  146.754980] ch341-uart ttyUSB0: ch341-uart converter now disconnected from ttyUSB0
[  146.754995] ch341 1-1:1.0: device disconnected
[  150.243591] usb 1-1: new full-speed USB device number 7 using xhci_hcd
[  150.392701] usb 1-1: New USB device found, idVendor=1a86, idProduct=7523
[  150.392704] usb 1-1: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[  150.392706] usb 1-1: Product: USB Serial
[  150.393341] ch341 1-1:1.0: ch341-uart converter detected
[  150.393695] usb 1-1: ch341-uart converter now attached to ttyUSB0
```

Thanks in advance.
Merry Christmas and enjoy the holidays.
Seb

seb-koch, 2021-01-07:
I found the root cause. It seems like a `sudo` doesn't give enough privileges to communicate with the interface. Once switched into root via `sudo su`it worked like a charm. You can close the issue and i suggest to change the documentation and state clearly that full root is necessary.
The docs aren't in here afaik, i'd be happy to alter the docs via a PR if this helps?

closed 2021-01-07 by seb-koch

RaymondMouthaan, 2021-02-05:
@seb-koch,

I am facing the same issue on one of my CC2652R usb sticks. However `sudo su` didn't work for me :( 

While pressing and holding the small button and putting the stick in the USB port, the led remains blinking. Then I run the command below with the given error:

```
root# ./cc2538-bsl.py -p /dev/ttyUSB0 -evw blink.bin
Opening port /dev/ttyUSB0, baud 500000
Reading data from blink.bin
Cannot auto-detect firmware filetype: Assuming .bin
Connecting to target...
ERROR: Timeout waiting for ACK/NACK after 'Synch (0x55 0x55)'
```

In this case I am just flashing the blink.bin, but same error occurs when flashing the .hex

Looks like the stick doesn't go to flash mode? Any idea how to fix this?

Thanks in advance,
Ray

VincentSC, 2021-12-11:
FYI: for me the `sudo su` trick *did* work.

zen2, 2021-12-11:
I think you can simply become root first:

`# sudo -i`  [enter the user password]
or
`# su -`                     [enter the root password]

VincentSC, 2021-12-11:
It is not about becoming root, but that the python-script does not work by running the script as root.

zen2, 2021-12-13:
> It is not about becoming root, but that the python-script does not work by running the script as root.

The script is working perfectly as root. It's not working when the script is launched via sudo.
In fact sudo restrict root environment by default so technically the script is launched as root but don't have access to everything that real user root can access. Than can be relaxed by sudo configuration or simply by becoming root with a full root environment like a real login result. That's the purpose of both command I posted.
These commands can be completed with '-c' option to add a command to be launched in resulting login shell.

VincentSC, 2021-12-13:
I meant "running as sudo" - matter of speak. It's a documentation problem then.

ovizii, 2022-06-09:
I've been scratching my head the whole time as SONOFF Zigbee 3.0 USB Dongle Plus does not have a button. I then figured out I'd have to open its metal case to get to the buttons? 

In the picture by itead there are two buttons: 
- boot button
- reset button

I assume you're all talking about the boot button?