electrolama/zig-a-zig-ah #23: Enhancement : Add RF amplifier CC2592

wojciechczyz, 2021-01-11:
Are you considering creating version with amplifier such as CC2592?

Hedda, 2021-01-13:
CC2592 + CC2652R works but such solution would be more expensive than using a CC2652P instead of CC2652R + CC2592. Read:

https://e2e.ti.com/support/wireless-connectivity/bluetooth/f/538/t/965346?CC2652P-Update-from-CC2640-CC2592-to-CC2652P

FYI, while "zzh-p" is not yet available the design use CC2652P instead which does have an internal RF PA (power-amplifier) integrated:

https://electrolama.com/projects/zig-a-zig-ah/#Flavours

 **"_zzh-p is the experimental version with CC2652P (built-in PA). Shares most of the same design with zzh except for the RF parts and will fit in a slightly larger case. It is a work in progress, sign up to the Electrolama mailing list to be notified of project updates and when kits/assembled units go on sale. A sneak peek:_"**

[image]

https://electrolama.com/projects/zig-a-zig-ah/#aside-ti-part-numbers

**_"CC2652P = CC2652R with a built-in RF PA. Not pin or firmware compatible with CC2652R/CC2652RB. zzh-p will use this._**"

omerk, 2021-01-22:
Thanks for the suggestion @wojciechczyz, as Hedda mentioned there is a new design (two, in fact) based on CC2652P that does have a PA built-in. Keep an eye on the repo/website (electrolama.com) 🙂

closed 2021-01-22 by omerk