electrolama/zig-a-zig-ah #24: zzh-p stick based CC2652P?

Hedda, 2021-01-22:
Noticed that Koenkk is now also hosting Z-Stack 3 coordinator firmware for "Electrolama zzhp" and "Electrolama zzhp-lite":

https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_3.x.0/bin

@omerk Any status updates on Electrolama's "zzh-p" and "zzhp-lite" products based on CC2652P?

https://electrolama.com/projects/zig-a-zig-ah/#Flavours

"**_zzh-p is the experimental version with CC2652P (built-in PA). Shares most of the same design with zzh except for the RF parts and will fit in a slightly larger case. It is a work in progress, sign up to the Electrolama mailing list to be notified of project updates and when kits/assembled units go on sale. A sneak peek:_**"

[image]

omerk, 2021-01-22:
Hi @Hedda,

There will be two versions zzh with CC2652P: zzhp and zzhp-lite. Both will use the same firmware, zzhp-lite will be module-based (PCB antenna) and zzhp will be a custom board similar to the first draft screenshot you linked above (external antenna). I've also managed to squeeze in a few more goodies in zzhp and will be updating the website soon.

Release date not set yet but it's _very_ soon.

Happy to send you a prototype as a thank you for your contributions to the documentation and keeping an eye on these issues, please drop me an email (support @ electrolama.com) with your address 🙂

closed 2021-01-22 by omerk

Hedda, 2021-03-24:
> Happy to send you a prototype as a thank you for your contributions to the documentation and keeping an eye on these issues, please drop me an email (support @ electrolama.com) with your address 🙂

Hi @omerk I can wait, but maybe you consider sending one to @zoic21 and/or @sarakha63 to test with https://github.com/zigpy/zigpy/pull/656 (ZGP)?